fossasia/pslab-firmware #40: Feature: SD Card support library

Dilu9218, 2019-04-07:
**Expected**: Firmware needs to support SD Card read/write functions in PSLab-v6 design

**Actual**: Current firmware doesn't have any functionality to support this feature

I would like to add required library files and I will refer this open source [FatFS](http://elm-chan.org/fsw/ff/00index_e.html) library which comes with a [GNU GPL licence](http://elm-chan.org/fsw/ff/doc/appnote.html#license). Since I don't have the device with an SD card mounted, I would try out the code with existing PSLab device and an external [SD Card Module](https://potentiallabs.com/cart/image/cache/catalog/Oct-2018%20components/Micro%20SD%20Card%20Module1-800x800.jpg) connected with SPI protocol. Once that is done, @CloudyPadmal, could you please check it with the new prototype and provide feedback?

CloudyPadmal, 2019-04-07:
Hello @Dilu9218 , Yes this seems like a possible approach. Please send us code in a PR to review. I'll check with my prototype. Thanks.

CloudyPadmal, 2019-04-07:
Just to define things clearly, following is the pins assigned;

- SPI Clock **RC3**
- SPI Data In **RA9**
- SPI Data Out **RA4**
- Chip Select **RB9**
- Switch **RC5**
- Status LED **RB15**

CloudyPadmal, 2019-07-29:
Will address in #46

closed 2019-07-29 by CloudyPadmal