fossasia/pslab-firmware #46: Upgrade firmware to support SD card logging

CloudyPadmal, 2019-07-29:


hkbiet, 2020-02-05:
Is this done ?

froscon2024, 2025-03-04:
The v6 hardware release have a coin cell battery holder. It would be great to be able to have the logging capability on sd card build into the firmware.
A use case would be: Configure the time into the pslab, disconnect pslab and use it as a modern voltmeter-multimeter with logging capabilities. The timestamps could help in later documentation of work that have been done with the pslab on the go.

bessman, 2025-03-04:
This is on our roadmap. It's technically already possible to do this if you're familiar with C programming; all the required features are already present in the firmware. However, they are not exposed via a nice API; that is something I would like to improve.