fossasia/pslab-firmware #47: Add http server code to communicate with PSLab over wifi using ESP

neel1998, 2019-08-08:
Currently, there is no functionality available to communicate with PSLab wirelessly. This issue aims to add server code for wifi communication

neel1998, 2019-08-12:
@CloudyPadmal now basic code is there... which other endpoints we aim to add to server?

neel1998, 2019-08-12:
Also wont we need a c library like other functionalities so we can connect esp module to pslab and also get voltage data from PSLab and broadcast them over wifi

CloudyPadmal, 2019-08-13:
Think about ESP chip as the same UART chip. UART chip has some internal firmware 
- to send data from outside to microcontroller and 
- to send data to outside from microcontroller

Functionality of ESP is the same. It will have an endpoint to write whatever the data we are sending to serial port and read from serial port and send whatever the data coming from microcontroller to outside. No need to do any data processing inside ESP. Only two endpoints are needed (Get and Post). But we can customize and have multiple endpoints to group calls and increase performance. But the basic idea is similar to what the UART chip is doing. ESP will be just an interface.

neel1998, 2019-08-13:
so will need to endpoints as you said... getData() and sendData().... on the ESP side... and from the PSLab android app we will call these endpoints when we want to communicate... for example to get version of PSLab... we will first send (POST request) 11 and 5 to sendData endpoint and then will make a GET Request to getData() to receive the data. Sounds right?

CloudyPadmal, 2019-08-13:
Yes. Sounds right.

neel1998, 2019-08-13:
So for now I will keep the post request contain simple json object with one parameter of data having the required byte string. Also from the proposed methods i think the 2nd one seems logical. Creating two flavors would add a lot of redundancy i believe. I will open an issue to add http communication handler in android app.

Hyperion101010, 2020-04-20:
Looks like this project https://github.com/fossasia/labs.fossasia.org/issues/286 is dependent in some way on this issue.

PR #48 (2019-08-11): feat:basic code for ESP server added
Fix #47