fossasia/pslab-firmware #105: Additional bus functions

nkpro2000sr, 2021-07-12:
* **SEND_SPI8_BULK**

   Currently we only have **SEND_SPI8** which send a single (8-bit) word. Sending array of words need sending repeated **SPI_HEADER** and **SEND_SPI8** before each word.  
   Like `[SPI_HEADER][SEND_SPI8][word_out_0][word_in_0][ACK][SPI_HEADER][SEND_SPI8][word_out_1][word_in_1][ACK]...`  
   By **SEND_SPI8_BULK** we can avoid sending repeated **SPI_HEADER** and **SEND_SPI8**.  
   Like `[SPI_HEADER][SEND_SPI8_BULK][count][word_out_0][word_in_0][word_out_1][word_in_1]...`

* **SEND_SPI16_BULK**

   Same as **SEND_SPI8_BULK** with (16-bit) word.

* **WRITE_SPI8_BULK**

   For writing only, we can also avoid `[word_in_\d+]`.  
   Like `[SPI_HEADER][WRITE_SPI8_BULK][count][word_0][word_1]...`.

* **WRITE_SPI16_BULK**

   Same as **WRITE_SPI8_BULK** with (16-bit) word.

* **READ_SPI8_BULK**

   For reading only, we can avoid `[word_out_\d+]`.  
   Like `[SPI_HEADER][READ_SPI8_BULK][count][word_0][word_1]...`.

* **READ_SPI16_BULK**

   Same as **READ_SPI8_BULK** with (16-bit) word.

* **I2C_SEND_BULK**

   Currently we only have **I2C_SEND_BURST** which send a single byte. Sending array of bytes need sending repeated **I2C_HEADER** and **I2C_SEND_BURST** before each byte.  
   Like `[I2C_HEADER][I2C_SEND_BURST][data_0][I2C_HEADER][I2C_SEND_BURST][data_1]...`  
   By **I2C_SEND_BULK** we can avoid sending repeated **I2C_HEADER** and **I2C_SEND_BURST**.  
   Like `[I2C_HEADER][I2C_SEND_BULK][count][data_0][data_1]...`  

* **I2C_READ**

   Currently we use **I2C_READ_MORE** and **I2C_READ_END** which reads byte by byte.  
   By using **I2C_READ** we can avoid sending repeated **I2C_HEADER**, **I2C_READ_MORE** and **I2C_READ_END**.  
   Like `[I2C_HEADER][I2C_READ][count][data_0][data_1]...`

CloudyPadmal, 2022-09-25:
Are `SEND` and `WRITE` operations the same?

bessman, 2023-09-15:
#148 adds full support for 8- and 16-bit SPI transactions.

closed 2023-09-15 by bessman