fossasia/pslab-firmware #182: I2C Device Scan doesn't work

AsCress, 2024-12-28:
If one tries to scan for a list of connected I2C devices using the `scan` function, using either _pslab-python_ or _pslab-android_, the PSLab device replies in affirmative to each and every address scanned for (see https://github.com/fossasia/pslab-android/issues/2595 in _pslab-android_).
I first noticed this issue in the android app, however, when I tried to scan using pslab-python, I got the same results.
Attaching a screenshot for your reference,

[image]

### Why did I create an issue in the firmware repository?
I remember when I used this feature with the legacy (2.0) firmware. It worked well both in the android app as well as using the python library.
Thus, it seems to me that the issue may be somewhere in the firmware, and not in _pslab-python_ or _pslab-android_.
@bessman Have you tested this before ? Has the I2C implementation in the firmware changed in anyway between the legacy (2.0) and the latest (3.0) firmware (I remember you telling me that it hasn't, I guess ?)
Nonetheless, I'd love to get this solved.

bessman, 2024-12-29:
This is related to a change in the return value from `I2C_CommandStart`. It used to return the `I2C2STATbits.ACKSTAT` bit along with the regular ACK byte. Simply readding the ACKSTAT bit to the return value did not work for me, however. I will keep investigating.

bessman, 2024-12-29:
I am concerned that this may be a hardware problem. Adding I2C2STATbits.ACKSTAT to the return value solves the problem when using v5 hardware. :thinking:

AsCress, 2024-12-29:
> I remember when I used this feature with the legacy (2.0) firmware. It worked well both in the android app as well as using the python library.

I think when I observed this I flashed the legacy firmware to the v6 board ( I never had the v5 board with me 😂 ).

bessman, 2024-12-29:
OK, I will try with legacy firmware.

bessman, 2024-12-29:
With legacy firmware I observe the same behavior on legacy and current hardware: I2C scan works once, then returns nothing.

AsCress, 2024-12-31:
@bessman You are right. I am observing the same behavior with the v6 hardware: 

[image]

Maybe I didn't notice this earlier because I never used the I2C scan in the app more than once even with the legacy firmware, and the first time it works well.

bessman, 2025-01-05:
I found the problem. It's partly a firmware problem, which was fixed by 6356eb2656f856327c9090bcbd6e404db78ae2f0, partly a problem in pslab-python's peripheral scanning.

When checking if a peripheral device is present, pslab-python does this:

[image]

1. START condition
2. ADDRESS
3. READ
4. Waits for peripheral ACK
5. STOP condition

This is not a legal I2C transaction. After the controller sends a READ command, it must read at least one byte of data from the addressed peripheral, followed by NACK and STOP. Sending STOP immediately after READ leaves the peripheral in an undefined state.

The correct way to do this is to send WRITE instead of READ, which it looks like pslab-android already does: https://github.com/fossasia/pslab-android/blob/15a487b69c9485e81349461dcd61e437771b49b6/app/src/main/java/io/pslab/communication/peripherals/I2C.java#L212C21-L212C26

So device scanning should work on Android with 6356eb2656f856327c9090bcbd6e404db78ae2f0. @AsCress can you confirm? If it works I will make a new firmware release.

AsCress, 2025-01-05:
Closing this as fixed by https://github.com/fossasia/pslab-firmware/commit/6356eb2656f856327c9090bcbd6e404db78ae2f0.

closed 2025-01-05 by AsCress