fossasia/pslab-firmware #189: Inculde bootloader in firmware release images

bessman, 2025-02-17:
The current release process automatically builds and attaches firmware to the release page. These firmware images includes only the application firmware, not the bootloader. If someone were to flash these images with a PICkit, the existing bootloader would be erased and the PSLab would no longer be able to enter bootloader mode.

It would be good to also build the bootloader and merge it into each of the release images, as documented [here](https://github.com/fossasia/pslab-bootloader/?tab=readme-ov-file#creating-a-combined-hex-file). Then the release images can be safely flashed with a PICkit.

froscon2024, 2025-03-04:
Is it possible to update the bootloader with such files? So the user could choose if the .hex firmware that is going to be installed should also update bootloader (when user have a tool to recover a broken bootloader).

bessman, 2025-03-04:
If I understand your question correctly, yes, flashing such a combined bootloader + firmware image to the board using a hardware programmer (such as a PICkit) will also update the bootloader.

bessman, 2025-03-04:
> The people now are flashing the firmware+bootloader combined file and only when using a external programmer, this file updates the bootloader?

Yes, that is correct. A hex-file is essentially a bunch of data with associated addresses. When flashing over USB, any data associated with an address outside the memory area reserved for the main application is ignored. Thus, in the combined bootloader + firmware image, the bootloader part is simply discarded when flashing over USB.

In contrast, when flashing with a hardware programmer, the entire hex-file is written to the device. When using a combined bootloader + firmware image, this will update the bootloader. When using a firmware-only image, it will remove the bootloader.

PR #197 (2025-04-04): chore: include bootloader in release images
Fixes #189

As @bessman mentioned in the issue:
The current release process automatically builds and attaches firmware images to the release page. These firmware images include only the application firmware, not the bootloader. If someone were to flash these images with a PICkit, the existing bootloader would be erased and the PSLab would no longer be able to enter bootloader mode.

So, the bootloader image is added to the application image.

Did it in the following steps-

1. Building the bootloader image:
* Cloned the bootloader repository inside t [... cut]
closed 2025-04-04 by bessman