fossasia/pslab-firmware #191: Firmware release 3.1.0 is broken

froscon2024, 2025-03-04:
The release 3.1.0 is broken. It does not work. No communication with the pslab v6 hardware is possible. The firmware file itself even look broken. Its way too small. The firmware release 3.0.4 and before had packed firmware files of over 60Kilobyte. Suddenly the firmware size dropped in release 3.1.0 and the firmware is broken.
Also the firmware filenames in the archives for unknown reason changed. On older releases the firmware files for v6 had also v6 in their names.
Please remove the broken firmware files. Those do not even respond on serial command and the python and the android software does not work.

bessman, 2025-03-04:
Hi @froscon2024, and thanks for reaching out!

I cannot reproduce this issue using the release files from https://github.com/fossasia/pslab-firmware/releases/tag/3.1.0. Using pslab-firmware.hex (md5sum 76a8cc412bc2da4cb13676f7f98976e1) from pslab-firmware_v6.zip (md5sum efe0fb2aad9331b97c8a8d8c496faabc), the board responds as expected with pslab-python:

    $ python -c "import pslab; print(pslab.ScienceLab().device.get_version())"
    PSLab V6

Could you provide more information about your environment, please?

- Do the md5sums for the files you downloaded match the above?
- What version of pslab-android and pslab-python did you try?
- What operating system / android version are you using?

> The firmware file itself even look broken. Its way too small. The firmware release 3.0.4 and before had packed firmware files of over 60Kilobyte. Suddenly the firmware size dropped in release 3.1.0 and the firmware is broken.

The reason for the smaller file size is that unlike previous releases, the images contain only the firmware application, not the bootloader. This issue is tracked in #189, but should not affect the function of the firmware images.

> Also the firmware filenames in the archives for unknown reason changed. On older releases the firmware files for v6 had also v6 in their names.

This is because the release process has been automated. Previously, I manually built and uploaded the releases myself, now we are using a github action to do it. The process is very slightly different from the one I used, resulting in the different file name. It should not affect the function of the firmware image.

froscon2024, 2025-03-04:
Ok, i understood now why it was not working for me. The main readme of the this pslab-firmware repo does not contain this information from the pslab-esp01-firmware repo:
https://github.com/fossasia/pslab-esp01-firmware?tab=readme-ov-file#preparing-the-pslab-board

> After flashing this firmware to the PSLab, it will no longer communicate over the USB port. To restore the ability to communicate with the PSLab over USB, flash the 'pslab-firmware_v6.hex' image instead (the PSLab always communicates over USB while in bootloader mode, regardless of which application firmware is loaded).

Maybe add it to the main pslab-firmware readme so that the people know that they have to make a choice if they want to use USB or the ESP-01. Both things in one firmware file is not possible with current firmware files.

I installed 3.1.0 v6 without esp-01 support and usb communication is working fine now.

closed 2025-03-04 by froscon2024